magicbug/Cloudlog #458: Power level via ADIF

ofadam, 2020-04-09:
Is there a way to import and surface power levels via ADIF? These show in the QRZ log after import, but not Cloudlog.

magicbug, 2020-04-09:
If they we're included in the ADIF file on import they will be in the database I just never exposed the data.

m0pcb, 2020-04-09:
I like the idea of logging and seeing power level too.

Iain AD5XI/M0PCB

magicbug, 2020-04-09:
I'll add an input to the "station" tab for now and show on the details of QSOs when clicked will get that done later this evening / tomorrow

magicbug, 2020-04-09:
Started addressing this request in commit https://github.com/magicbug/Cloudlog/commit/625f828bd5b8e015f4129d0af68ebd29b12fb189 just need to add the input box next

magicbug, 2020-04-13:
This is now fully addressed

ofadam, 2020-04-14:
Is this retroactive for any ASIF imports, or do I need to reimport? Does the field only support raw numbers or "W" for watt on the end?

magicbug, 2020-04-14:
ADIF spec is power level in watts (number only), if adifs with this data has been imported & was valid then the data will be there.

closed 2020-04-14 by magicbug