magicbug/Cloudlog #465: LOTW download reset start date

pa3ang, 2020-04-17:
Peter,

how can i download all LOTW confirmations starting from 2001-01-06 . When doing the ADIF upload out of HRD, i'm missing all the LOTW verified QSO and when downloading in Cloudlog it's fetching only some last QSOs.

Regards,
Johan, PA3ANG

magicbug, 2020-04-20:
Hi @pa3ang,

At the moment it's done by checking the log to see when the last confirmation was guess better option might be adding an option to download all via a checkbox?

in Controller lotw.php
``
$lotw_last_qsl_date = date('Y-m-d', strtotime($this->logbook_model->lotw_last_qsl_date()));
``

pa3ang, 2020-04-20:
> Hi @pa3ang [1],

Hi Peter, 

Yes could be a good candidate to be added as more people would like to
migrate to cloudlog with their current logbook solouting. 

I downloaded all qso in the mean time by tweaking this line 'hard coded'
as a one off. 

//$lotw_last_qsl_date = date('Y-m-d',
strtotime($this->logbook_model->lotw_last_qsl_date()));
$lotw_last_qsl_date = "2000-01-01";

73 Johan 

 

Links:
------
[1] https://github.com/pa3ang
[2]
https://github.com/magicbug/Cloudlog/issues/465#issuecomment-616466598
[3]
https://github.com/

magicbug, 2020-04-20:
Changed title so its more obvious and added feature tag :)

commit ac3b0475 (2020-04-25): Merge pull request #479 from AndreasK79/lotw_report_datepicker

Added datepicker for LoTW report download. Fixes #465.
commit adbc7f68 (2020-04-25): Added datepicker for LoTW report download. Fixes #465.
closed 2020-04-25 by magicbug