magicbug/Cloudlog #470: "Any band" support for gridsquare map

M0LTE, 2020-04-20:
**Is your feature request related to a problem? Please describe.**
I can't see a map of all the grids I've worked.

**Describe the solution you'd like**

The following works: 
https://cloudlog.m0lte.uk/index.php/gridsquares/band/20m

It would be nice if the following (or similar) worked the same as above:
https://cloudlog.m0lte.uk/index.php/gridsquares/band/all

magicbug, 2020-04-20:
So ALL would be everything in the log? or you thinking narrowing it down 

- HF
- VHF
- UHF

etc?

M0LTE, 2020-04-20:
All is what I had in mind.

Could also accept HF/VHF/UHF as well I guess?

Would be nice to have a checkbox to be able to show or hide numbered grids irrespective of zoom level too.

closed 2020-04-25 by magicbug

M0LTE, 2020-04-25:
Legend :)

It's ace. Thanks.

hairlesshobo, 2020-12-06:
> Would be nice to have a checkbox to be able to show or hide numbered grids irrespective of zoom level too.

Was this also added when you added the "all" drop down selection, or would I need to create a new feature request for it. Currently, I have to zoom in much more than I would prefer in order to be able to see numbered grids. 

Thanks!

AndreasK79, 2020-12-06:
This was not added. I am also interested in this, but I'm not sure how to control the zoom and the displaying of the gridsquares.

AndreasK79, 2020-12-09:
@hairlesshobo please create a new issue for this feature.

hairlesshobo, 2020-12-09:
@AndreasK79  will do!

commit 949c421a (2020-04-25): Added all to the gridsquare select. Fixes #470.
commit de93daa3 (2020-04-25): Merge pull request #477 from AndreasK79/all_added_to_gridsquare_map

Added all to the gridsquare select. Fixes #470.