magicbug/Cloudlog #480: LoTW import broken

andrewbnortham, 2020-04-26:
Line 1143 in application/models/logbook_model.php needs a change, maybe something like this:    function import($record, $station_id = "0", $skipDuplicate**=F**, $markLotw**=T**, $dxccAdif**=T**) 

**Describe the bug**
When importing from LoTW, using either method, CloudLog will error when it has a record to Insert. 

An uncaught Exception was encountered
Type: ArgumentCountError

Message: Too few arguments to function Logbook_model::import(), 1 passed in /var/www/html/Cloudlog/application/controllers/Lotw.php on line 66 and exactly 5 expected

Filename: /var/www/html/Cloudlog/application/models/Logbook_model.php

Line Number: 1143

There was a change made 3 weeks ago to application/models/logbook_model.php on line 1143 and it now required 5 variables. When its called from application/controllers/lotw.php only I variable is passed.

magicbug, 2020-04-26:
Weird, just tried here via the "pull" feature and no error?

Running the latest codebase as of today?

AndreasK79, 2020-04-26:
The pull feature works, but when the qso is not found, the controller tries to import the qso, which I broke with my PR a few weeks back. The same thing happened with the api.

`if($status == "No Match") {
	// Create the Entry
	$this->logbook_model->import($record);`

I noticed this when I made the modifications to the LoTW-ctronller the other day. I haven't gotten around to a fix yet.

@magicbug I know you fixed the api with import($record, 0, NULL, NULL, NULL). Should the same thing be done here?

andrewbnortham, 2020-04-26:
Did you have any QSO's to ADD, or just to update? It only errors on the ADD, not the update. 

I had a almost empty database I was trying to import all of my previous contacts to.

Codebase update last was yesterday.

magicbug, 2020-04-26:
@AndreasK79 ah forgot that it will try to create the QSO, think we need to remove that feature, to be honest, seems unwise and also could break being assigned to a station profile. LoTW imports should only be marking things as "matched" but not creating a contact if it doesn't exist or that's my current feeling on the matter.

andrewbnortham, 2020-04-26:
I dont have a preference either way. I did simplify my process. WSJT-X QSO, Uploaded using TQSL. Wait for LoTW to process, perform LoTW download and it adds the QSO. 

I can import into CouldLod from WSJT-X, and download from LoTW. Seemed redundant at the time, but may explain some issues with awards not showing correctly...

andrewbnortham, 2020-04-28:
I have been troubleshooting why WAS wasnt working for my HF contacts, but has been working fine for my satellite contacts and some 6m. When the contact is imported directly from LoTW, it includes the State. Importing from TQSL does not include the state, and the update does not add it. 

One vote for keeping this feature to add missed contacts.

commit a9c957e6 (2020-04-28): Added fix for import of not found LoTW matches. Fixes #480.
commit e78fdf64 (2020-04-28): Merge pull request #482 from AndreasK79/fix_lotw_import

Added fix for import of not found LoTW matches. Fixes #480.
closed 2020-04-28 by magicbug