magicbug/Cloudlog #483: Adjustable QSO window layout

KeyOz, 2020-04-30:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
It would be good if you could make the layout of the QSO input adjustable so that you can use the full monitor area with large monitors. It would be nice to have 3 columns next to each other. So you would have a larger map to view.

myzinsky, 2020-05-06:
I also think that at least the frequency should be also in the front tab. Moreover, the editable contest fields on the front tab would be nice!

magicbug, 2020-05-08:
I get that configuration of the QSO windows probably high up on lots of users, this isn't something that is currently on the road map, I'll be making some changes soon, that will provide better choices of QSO boxes depending on the operating style.

However if users want further options beyond that then I'd suggest they edit the code to meet their own requirements.

closed 2020-05-08 by magicbug