magicbug/Cloudlog #488: qrz.com lookup does not work

myzinsky, 2020-05-02:
**Describe the bug**
The data lookup of QRZ.com does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Live QSO
2. Add call sign
3. Click on suggested call sign
4. No name or locator is filled out

**Expected behavior**
QRZ.com Lookup should work

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Chrome
 - Version 81.0.4044.113

magicbug, 2020-05-02:
QRZ works as follows

- Input QSO
- Move out the field (usually TAB)
- Fields populated

myzinsky, 2020-05-03:
Okay I narrowed it down. In Firefox on Windows and Safari on Mac it does work but in Chrome on Windows it does not work. Something to do that I can help you to reproduce?

magicbug, 2020-05-03:
Weird, I can check firefox, just a thought tried logging in back in on Firefox to see if it makes any difference? if not it will be a case of seeing if the console log is throwing errors.

magicbug, 2020-05-08:
Finally had time to check on firefox here (lastest version no plugins) logged into Cloudlog, typed in callsign "M0VFC" tabbed out of the field and information was populated.

So this is likely to be one of a few things
- Session Key for QRZ is expiring faster than it should (This can be tested by logging in and out of Cloudlog)
- A configuration issue with the server, this isn't so easy to test unless I create some debug tools so you can quickly call QRZ and it will return errors if its not the first then I can add that.

myzinsky, 2020-05-08:
Firefox works fine Chrome on windows Is making the issue.

magicbug, 2020-05-09:
Do you have any plugins installed if you have the developer tool bar open  control + shift + I do you see any errors? this can only be plugin related and a bit beyond the cope of  code support.

Chrome on Windows is my daily driver so if that was broken I'd quickly notice, I'd disable all plugins and work your way through to whatever breaks it.

closed 2020-05-13 by magicbug