magicbug/Cloudlog #493: Edit / delete QSO : Trying to get property of non-object

m0boy, 2020-05-10:
Multiple PHP notices when trying to edit/delete QSO, 
A PHP Error was encountered

Severity: Notice

Message: Trying to get property of non-object

Filename: qso/edit.php

Line Number: 49
">
End Date/Time.. same for multiple actions, Line numbers: 49,54, 60,63, 121,126,137,146,158,163,168,195,225,331,497

A clear and concise description of what the bug is.
Add dummy QSO, then click the QSO, Edit QSO then Delete QSO.

[image]
[image]

A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Ubuntu]
 - Browser [Firefox,]
 - Version [latest]

**Smartphone (please complete the following information):**
 - Device: [iPhoneXSMax]
 - OS: [Latest IOS]
 - Browser [Safari]
 - Version [built in]

**Additional context**
Add any other context about the problem here.

magicbug, 2020-05-10:
Something not right there, what was the contents of the dummy QSO?

magicbug, 2020-05-10:
Also worth asking is this the latest version from git or older

m0boy, 2020-05-10:
Hi, Was having this issue since first time installed cloudlog about 6mo ago. Content of QSO is irrelevant, is doing it with any qso. Did not pay attention to it as I was able to delete the QSO if needed so. However good habbits to report issues. 
Updated cloudlog 2 weeks ago.

magicbug, 2020-05-10:
Yeah that's not right, I'm guessing country files updated.

I'll go through the errors see if I can figure out what's up but that's not normal behaviour

m0boy, 2020-05-10:
[image]
Might be useful :)

magicbug, 2020-05-10:
That is rather a bazaar issue looking at the code errors report to fields being echoed so if this was a widespread issue I'd have seen it, or I'd hope so.

Can you do a more recent code update, not convinced it will make any difference but the rare chance it might? thats basically saying nothings coming back from the server which feels unlikely.

magicbug, 2020-05-10:
Going to tag @dg9vh and @AndreasK79 just on the off chance they have an idea.

m0boy, 2020-05-10:
Update didn't help. Something really stuck there, created a fresh install in a new folder with new database: 
[image]

Please note that I had this problem very first day on my first install.

dg9vh, 2020-05-10:
Hi,
as I can see, this is only(!) a notice-error-message, so it could be ignored and the webserver itself should be configured to suppress this messages...

The cause of this notice could be, that echo $qso->COL_PRIMARY_KEY; is operating on a null-value in $qso, so getting the attribute COL_PRIMARY_KEY of this would end in such an errormessage.

Question is, why $qso would be null?

AndreasK79, 2020-05-10:
Can't say I understand what's going on here, but since a new install works, something must have happened with the old install. @m0boy do you get the same error if you switch to the old databasen in the new folder?

m0boy, 2020-05-10:
The only occurence I managed to make it work is the one above. Did 3 reinstalls since and all coming up with the error. 3 different databases. files uploaded through FTP.
https://logbook.m0boy.uk/index.php please feel free to log in with defaults (m0abc) and test

buhalo1, 2020-05-23:
Just installed cloudbook and having the same issue. 
$config['directory'] = "/logbook";
$config['base_url']	= 'https://callsign.radio1234.org/logbook/';

dg9vh, 2020-05-23:
I think you doubled the information... try to let config[directory] empty...

buhalo1, 2020-05-24:
Tried to make clean install. Got the following screen on first run.

[image]

dg9vh, 2020-05-24:
What happend after login and creating an own user and setting station profile?

dg9vh, 2020-05-24:
By the way: this is just a notice-message that could be ignored... you should configure your php.ini to suppress such notices on production-services.

buhalo1, 2020-05-24:
Those notices dissapeared on page refresh. 
QSO delete errors are still there.

magicbug, 2020-05-24:
This is definitely relating to a configuration with a server as if it was the code it would have been reported by everyone using it., hiding those errors doesn't fix the problem they shouldn't exist in the first place, 

My first suggestion would be to use a newer PHP version, development-wise I'm testing on PHP 7.3 & 7.4.. 7.3 is certainly stable. 7.0.33 is end of life since the start of 2019

buhalo1, 2020-06-12:
After successful QSO removal it tries to get back to previous page which is the QSO edit page, then it tries to populate fields on this page by quering the content of QSO fields from database. The error is that it tries to load data for QSO that just been removed.

closed 2020-09-30 by magicbug