magicbug/Cloudlog #498: 404 Not found

ea7bj, 2020-05-15:
after entering m0abc/demo,  I receive a "404 page not found"

The content of $ config ['base_url'] is

$ config ['base_url'] = 'http://192.168.1.110/Cloudlog';

It is the local IP of the raspberry.

any suggestion ?

73 EA7BJ Jose

magicbug, 2020-05-15:
If your getting 404 either you have a configuration issue with I'm guessing apache or your folder isn't in the correct location, did setup function? i

ea7bj, 2020-05-16:
Yes, the configuration worked fine, then delete the /install folder and put in the browser the local ip of the raspberry /Cloudlog, it asks me to identify myself, I enter m0abc/demo and the next page is ... 404 Not found.
I use nginx.
Curiously, without touching anything I managed to enter after many attempts, I was able to create a new user, delete the user m0abc, everything seemed to be working fine, but this morning error 404 appears again

magicbug, 2020-05-17:
OK, not sure how best to provide support on this, it would be better suited for the forum https://forum.cloudlog.co.uk as its not a bug with the codebase but local server, however, https://stackoverflow.com/questions/45633900/nginx-server-configuration-for-codeigniter might solve your question, else I'd recommend posting on the forum and seeing if anyone can help.

closed 2020-05-17 by magicbug

konaya, 2022-05-02:
I had the same issue, also running nginx. I'm wildly guessing it's because the `/index.php/etc/etc` scheme isn't very portable as-is between different web servers, or possibly doesn't work when running php-fpm as you would with nginx.

Luckily, it's pretty simple to fix. Just change `$config['index_page'] = 'index.php';` to `$config['index_page'] = 'index.php?';` and you're all set! Perhaps this warrants a footnote in the documentation?