magicbug/Cloudlog #501: Hint on CloudLogCatQt

myzinsky, 2020-05-17:
Hi, 

Since I'm a Mac user. I created a cross plattform tool for the remote CAT control that works together with FlRig: https://github.com/myzinsky/CloudLogCatQt

It would be nice if you could also link it in the right places in the readme and the wiki. 

vy 73 de DL9MJ, Matthias

magicbug, 2020-05-17:
Fantastic, I'm happy to get it added to the wiki, I've cleaned up the readme making it a little less chaotic.

magicbug, 2020-05-18:
@myzinsky Thanks for building this tool, I've added it to the wiki and given it a mention on the @cloudloghq twitter account! 👍

closed 2020-05-18 by magicbug

myzinsky, 2020-05-18:
Thx! Next project is a cross plattform app for pc, mac, android, linux as offline logbook for SOTA and outdoor activities with no LTE that pushes automatically the QSOs to cloudlog when back to Wifi or LTE ;-)

magicbug, 2020-05-18:
Shout if you need anything from the API, sounds like local app with database that kicks the API when connections back so that should all be available

myzinsky, 2020-05-18:
Yes I will do certainly. Is the documentation in the wiki complete? Or should I look in the source code?

magicbug, 2020-05-18:
90% complete always check source in case i added something