magicbug/Cloudlog #509: Post QSO feature: Date/Time

timkirby, 2020-05-22:
Entering Post QSO mode - the date and time seem to default to an old date/time (probably when you last used the feature). This seemed to start happening fairly recently - before which date/time would default to the current date and time.

magicbug, 2020-05-22:
Yeah this relates to #463 looks like the session needs to time out faster since @AndreasK79 added the option for it to hold the setting

AndreasK79, 2020-05-25:
Well, I just used the session that was there. Never thought about timing it out, but I guess adding the timestamp and diffchecking it before setting it, would take care of that problem. Now the only question is, how long should the timeout be?

AndreasK79, 2020-06-04:
Added #531. Resets date and time if the saved one was set more than 24 hours ago.

closed 2020-06-04 by magicbug