magicbug/Cloudlog #511: QSO detail window oversized after search

Ciemon, 2020-05-24:
Whilst testing the new QRB feature I searched for a recent QSO and noticed that the QSO detail window, accessed by clicking a link from the search results is expanded to a large percentage of screen width rather than the usual size.

[image]

magicbug, 2020-05-24:
Yep looks like i didnt do sizing on that bug needs addressed

Ciemon, 2020-05-24:
If it's a matter of copying the code from the main qso window to this one, I could do that if I knew which files it was.

magicbug, 2020-05-24:
Changed the code but it made zero difference so there's something else going on, will need to look when more brain cells are firing

AndreasK79, 2020-09-26:
This will no longer be an issue when PR #629 is merged. Bootstrapdialog has now been implemented.

closed 2020-09-28 by magicbug