magicbug/Cloudlog #512: Using COL_NOTES or COL_NOTES_INTL for internal notes

dg9vh, 2020-05-24:
**Is your feature request related to a problem? Please describe.**
Some users complained about the visibility of the infos filled in the comment's field for public and QSL-outout.

**Describe the solution you'd like**
Adding a Textbox that uses COL_NOTES that is already in the log-entry-table as column and that is only visible for logged in users within the QSO-edit and QSO-details-view, this problem would be solved. So users can do their comments not public visible and we have no further trouble on that.

Notes is a long-text-field that can contain further more characters than the comments field.

**Describe alternatives you've considered**
No cheaper alternative seen.

**Additional context**
Fields are also specified in ADIF-specs, so we can consider exporting the infos also in adif...

magicbug, 2020-05-24:
Tbh comment field is used per spec, the problem is some seem to perhaps leave unsavoury comments about those they are working in the field.

I'd support the development of a popup box on the QSO entry area where people can leave longer notes about the QSO, I'd suggest popup as it would give a decent window for note writing

magicbug, 2020-05-24:
As for which field always go with non INTL fields so **COL_NOTES**

dg9vh, 2020-05-24:
I would also prefer to have te col_notes-field used for this...

If anyone of our novice programmers would take the challenge?

I think using a separate tab in QSO-entry-mode with a nice big textarea would solve the problem. Using a Popup on QSO-Details (if clicked on the callsign in logbook-table) would also be nice and saves space on the window.

magicbug, 2020-06-04:
Implemented into the master code base

closed 2020-06-04 by magicbug