magicbug/Cloudlog #513: Error 1054 Unknown column qrzapikey in field list

ea5wa, 2020-05-24:
After updating my cloudlog (installed on a wamp server) I have an error message after first new entry

A Database Error Ocurred
Error Number: 1054
Unknown column 'qrzapikey' in 'field' list
File Name: DB_driver.php
Line Number: 691

dg9vh, 2020-05-24:
Do you have access to your database via phpmysqladmin or something similar?

ea5wa, 2020-05-24:
Yes, I have it

dg9vh, 2020-05-24:
What version do you get after
SELECT * FROM `migrations`
?

ea5wa, 2020-05-24:
> What version do you get after
> SELECT * FROM `migrations`
> ?

35

dg9vh, 2020-05-24:
This is far away from actual version... are you sure, your webserver has full access (read and write) to the migration-directory?

what is the latest file you have in /application/migration ?

ea5wa, 2020-05-24:
> This is far away from actual version... are you sure, your webserver has full access (read and write) to the migration-directory?
> 
> what is the latest file you have in /application/migrations ?

Latest file is from yesterday....
028_logbooks_setkeys.php
and some other files at the same hour

dg9vh, 2020-05-24:
I mean, what is the highest prefix-number in this folder? should be 040_...

If not, then your update is not completely with all files... You should download new ZIP-file from github (or do a git pull) to get all actual files.

ea5wa, 2020-05-24:
> I mean, what is the highest prefix-number in this folder? should be 040_...
> 
> If not, then your update is not completely with all files... You should download new ZIP-file from github (or do a git pull) to get all actual files.

yes, you're right, 40 is the latest

magicbug, 2020-05-24:
If you still have that error and you are seeing up to number 40 in migrations and you have error reporting turned on if you go `index.php/migrate` it will run the migration system and report any errors.

ea5wa, 2020-05-25:
Finally I went back to the old version.....

This afternoon I'll try it again

Thanks

ea5wa, 2020-05-25:
I tried again, and I have this error on index.php/migrate:

A Database Error Occurred
Error Number: 1050

Table 'eqsl_images' already exists

CREATE TABLE `eQSL_images` ( `id` INT(5) UNSIGNED NOT NULL AUTO_INCREMENT, `qso_id` VARCHAR(250) NOT NULL, `image_file` VARCHAR(250) NOT NULL, `modified` timestamp NULL, CONSTRAINT `pk_eQSL_images` PRIMARY KEY(`id`) ) DEFAULT CHARACTER SET = utf8mb4 COLLATE = utf8mb4_general_ci

Filename: C:/wamp/www/cloudlog/system/database/DB_driver.php

Line Number: 691

Any ideas?

magicbug, 2020-05-25:
When you restored the code yesterday did you restore the database too? eQSL_images  is being reported as already existing there for would suggest, if your rolling back and forth with the code like this and not rolling the database back at the same time this will be why your having such big issues all the time.

Only thing you can do is delete what its complaining about which is the eQSL_images table but I foresee other problems

ea5wa, 2020-05-25:
Yes, I've deleted and now I have other error messages....

ea5wa, 2020-05-25:
My suggestion, there should be am easier way to update.... Now seems a Logbook for Masters of MySQL

ea5wa, 2020-05-25:
> When you restored the code yesterday did you restore the database too? eQSL_images is being reported as already existing there for would suggest, if your rolling back and forth with the code like [...]

I've done what you told me long time ago....

Restore on the www folder the "aplication/config" folder

magicbug, 2020-05-25:
Feel free to submit a PR with the new improved update system.

magicbug, 2020-05-25:
You cant update the codebase and roll it back without restoring the database, there is limits of what the web script can do..

ea5wa, 2020-05-25:
Ok. What can I do now.... 

1. Paste the downloaded system
2. Restore application/config folder
3. Restore the database? Where is located?

ea5wa, 2020-05-25:
> Feel free to submit a PR with the new improved update system.

I wish I knew how to do it....

magicbug, 2020-06-03:
based on Twitter DM i believe this is solved so closing the issue

closed 2020-06-03 by magicbug