magicbug/Cloudlog #515: Adding COTA to the awards section

myzinsky, 2020-05-25:
In the QSO awards section we have already IOTA and SOTA. Recently I figured out the nice idea of COTA -- Castles on The Air (https://www.cotagroup.org/cotagroup/) which I think would be also nice to be added here. 

I would give it a try with the implementation. My approach would be to grep for IOTA and SOTA in the code and add COTA in the same places as well. Would there be another advise from your side?

magicbug, 2020-05-25:
Should work by borrowing bits of code, be careful to explain what this are doing in the code then if there's problems is easier to troubleshoot.

myzinsky, 2020-05-25:
Quick question: 

./application/views/adif/data/exportall.php
./application/views/adif/data/exportsat.php

are not really readable because most of the code is in one line. Is there a specific reason for that?

magicbug, 2020-05-25:
It's due to the fact if you space it out you start adding spacing to the exported file

myzinsky, 2020-05-25:
Opened PR (#518)

daccle, 2020-11-16:
I possibly solved by #688 as COTA seem to follow the MY_SIG and MY_SIG_INFO pattern as, POTA and WWFF.

daccle, 2020-11-16:
See my comment on  #705. I'd like to avoid adding COTA, POTA, WWFF to the AWARD section each on his own, since all of them are based on SIG/SIG_INFO and MY_SIG/MY_SIG_INFO. There should be a more general solution to have them all.

daccle, 2020-11-16:
BTW: the issue with exportall.php being hard to read is improved by #699

closed 2020-12-11 by magicbug