magicbug/Cloudlog #516: Easier way to update

ea5wa, 2020-05-25:
From my point of view, updating cloudlog is quite difficult for normal people.

There should be a way to update "automatically"

Everytime I've updated I've finished with errors....

magicbug, 2020-05-25:
I'd like to know how you think this can be achieved based on the current tooling and resources available.

magicbug, 2020-05-25:
Current updating system is as follows if you use git

Move to the directory type "git pull" back sure you have backed up your database which is just good practice and has to happen for all systems then log in the system then runs the migration.. so far over 100 installs haven't failed.

If you don't use git, then you download the files again, update the files in /config so they match the previous install then again login.

I could spent 2 months building a better upgrade system but I don't have WordPress resources available and it would mean a full halt to the project.

nr0q, 2020-05-25:
I haven't been able to do a git pull for a while, I keep getting

[mchambers@www Cloudlog]$ git pull
From github.com:magicbug/Cloudlog
   597c898..0c11f4e  master     -> origin/master
Updating 0628e3b..0c11f4e
error: Your local changes to the following files would be overwritten by
merge:
        README.md
        application/config/cloudlog.php
        application/config/config.sample.php
        application/config/migration.php
        application/controllers/Adif.php
        application/controllers/Api.php
        application/controllers/Awards.php
        application/controllers/Eqsl.php
        application/controllers/Gridsquares.php
        application/controllers/Lotw.php
        application/controllers/Update.php
        application/controllers/User.php
        application/libraries/Adif_parser.php
        application/libraries/Eqsl.php
        application/libraries/Frequency.php
        application/libraries/Hamqth.php
        application/libraries/Qra.php
        application/models/Adif_data.php
        application/models/Cat.php
        application/models/Cq.php
        application/models/Dxcc.php
        application/models/Gridsquares_model.php
        application/models/Stations.php
        application/models/Was.php
        application/views/adif/data/exportall.php
        application/views/adif/data/exportsat.php
        application/views/adif/import.php
        application/views/awards/cq/index.php
        application/views/awards/dxcc/index.php
        application/views/awards/nav_bar.php
        application/views/awards/was/index.php
        application/views/dashboard/index.php
        application/views/eqsl/import.php
        application/views/gridsquares/index.php
        application/views/interface_assets/footer.php
        application/views/interface_assets/header.php
        application/views/interface_assets/mini_header.php
        application/views/lotw/import.php
        application/views/search/result_search.php
        application/views/station_profile/create.php
        application/views/station_profile/edit.php
        application/views/user/edit.php
        application/views/view_log/partial/log.php
        application/views/view_log/qso.php
        assets/css/general.css
        assets/js/leaflet/L.MaidenheadColoured.js
        assets/js/radiohelpers.js
Please, commit your changes or stash them before you can merge.
Aborting
[mchambers@www Cloudlog]$

I'm not yet fluent in git to know how to tell it to go ahead and overwrite,
I have my whole /config dir backed up just in case

Matthew Chambers, CBT, NR0Q
Owner/Engineer
*M Chambers Communications Engineering LLC*
PO BOX 855, Moberly, MO 65270
Mobile (660)415-5620
www.mchambersradio.com

magicbug, 2020-05-25:
@nr0q better suited to a new issue although this isn't anything to do with the code base.

You have of course made changes to cloudlog.php in the config folder 

git stash 
git pull

ea5wa, 2020-05-25:
> Current updating system is as follows if you use git
>
> Move to the directory type "git pull" back sure you have backed up your database which is just good practice and has to happen for all systems [...]

In bold, what I've done..... Result = errors

ea5wa, 2020-05-25:
I'm giving a try to git on my windows server....

I've installed git, I've configured with user and mail.

I've moved with git terminal to my www/cloudlog on wamp, type "git pull" and I have a error message:  is not a git repository 

I need some help

dg9vh, 2020-05-25:
This only works, if you initially created your copy with git clone... if it is based on something else, it does not work this way...

magicbug, 2020-05-25:
> I need some help

Stop creating multiple issues then and stick to one so people might be able to help.

closed 2020-05-25 by magicbug