magicbug/Cloudlog #517: QRZ.com - no QSOs to upload found

dg9vh, 2020-05-25:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Although I have some edited QSOs in my profile, no QSO will be uploaded...

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Admin
2. Click on QRZ.com Export
3. Click on export in first row
4. See error

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 home 64
 - Browser Chrome
 - Version 83.0.4103.61

AndreasK79, 2020-05-25:
Hmm. After looking through the code, I think I know why, but not how to fix it.

At controllers/qrz.php line 67, I use a counter to count successful uploaded QSOs. If uploading these QSOs gives an error, they will not be counted, and not marked as uploaded.

If you can var_dump $content in logbook_model.php at line 377 ($content = curl_exec($ch)), that will probably give an answer.

dg9vh, 2020-05-25:
Doing this ends in an endless loop:
```
     curl_setopt( $ch, CURLOPT_FOLLOWLOCATION, 1);
      curl_setopt( $ch, CURLOPT_HEADER, 0);
      curl_setopt( $ch, CURLOPT_RETURNTRANSFER, true);

      $content = curl_exec($ch);
var_dump($content);
      if ($content){
          if (stristr($content,'RESULT=OK')) {
            return true;
          }
...
```

AndreasK79, 2020-05-25:
I guess you need to open devtools before you click on upload button. Then click on network and upload_station. Click response and see if the content of the variable was dumped there.

The reason for the dump, is that the return for the js ajax call is not returned correctly.

dl6nm, 2020-05-25:
I'm new to Cloudlog and I have the same problem on exporting to QRZ.com.

The response for `var_dump($content);` is as follows:

```
string(94) "STATUS=AUTH&REASON=user does not have a valid QRZ subscription to use this function
&EXTENDED="
[...]
string(94) "STATUS=AUTH&REASON=user does not have a valid QRZ subscription to use this function
&EXTENDED="
{"status":"Error","info":"Error, no QSOs to upload found"}
```

dg9vh, 2020-05-25:
```
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
string(69) "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED="
{"status":"Error","info":"Error, no QSOs to upload found"}
```

dg9vh, 2020-05-25:
Seems that it wants to add instead of update...

dg9vh, 2020-05-25:
> I'm new to Cloudlog and I have the same problem on exporting to QRZ.com.
>
> The response for `var_dump($content);` is as follows:
>
> ```
> string(94) "STATUS=AUTH&REASON=user does not have a valid QRZ [...]

Your's looks like you do not have an API-Key in your station-profile.

dg9vh, 2020-05-25:
What happend to my QSOs that they got the modify-flag is that I changed QSL-send-state to R(equested)... Maybe this change does not reflect in changeing the QSO for QRZ.com?

Are QSL-states being transported to QRZ.com?

AndreasK79, 2020-05-25:
@dg9vh Seems like I need to specify replace as an option for the modified QSOs.

dl6nm, 2020-05-25:
@dg9vh: Thanks for your reply. Hopefully I'm not hijacking your Issue. An API-Key I've provided in my station-profile, copied from the setting page on QRZ.com. Should I open a new issue on this?

AndreasK79, 2020-05-25:
@dg9vh I mark an edit as modified when saved, if an QRZ API Key exists. QSL-states are being transported to QRZ. Check logbook_model.php -> function create_adif_from_data.

dg9vh, 2020-05-25:
@dl6nm last for this: do you have a QRZ.com-subscription that allows usage of logbook?

dl6nm, 2020-05-25:
@dg9vh No, i haven't any subscriptions at QRZ.com. I'll have a look on this.
Maybe a better description or detailed error message especially for new users may help.

dg9vh, 2020-05-25:
@dl6nm ok, this is the source of your problem: you need a (paid) subscription (for example XML-Subscriber) to use this service...

@AndreasK79 by the way examining http://adif.org.uk/310/ADIF_310.htm#QSLSent_Enumeration I figured out another problem we have in QSL-printing :-) But this is "andere Baustelle"...

I can't figure out the problem in this case here...

AndreasK79, 2020-05-26:
@dg9vh I need to rewrite the code to send OPTION=REPLACE when uploading modified QSO.
From the API:
[image]

AndreasK79, 2020-05-26:
@dl6nm Yes, I agree with presenting the errors. The problem is that one call for one QSO is made, so the list of errors might be long.

@magicbug Any suggestions on how to log error messages and present them? One way I see, is to create a logging table, and insert into that. Then we can have a button that displays them in a dialogue.

dl6nm, 2020-05-26:
@AndreasK79 One solution could be logging erros to the logfile. This was the first place I've taken a look into. I've set the `log_threshold` to `4`, but I could not found anything regarding to my problem.

I've created a new feature request #522 for this.

magicbug, 2020-05-26:
Agreed @dl6nm all errors need to be returned to the user, I'd suggest if its API related then onscreen but Codeigniter does have log system build in [https://codeigniter.com/userguide3/general/errors.html](https://codeigniter.com/userguide3/general/errors.html) this seems a valid thing to push forward with.

AndreasK79, 2020-06-04:
I submitted #530 now. This will provide the necessary replace option when uploading modified QSOs. I will look into error logging later.

dg9vh, 2020-06-04:
With last merge, this issue is solved, i close it :-)

Best 73 and good work @AndreasK79

closed 2020-06-04 by dg9vh