magicbug/Cloudlog #522: Improve (error and debug) logging to easier get problems solved

dl6nm, 2020-05-26:
Regarding to #517 and #520, there is a gap to easy find out the source of problems.

As a user I'd expect useful messages on errors in the frontend and also in a logfile, if I activate the error logging feature.

magicbug, 2020-05-26:
Agree all errors should logged via https://codeigniter.com/userguide3/general/errors.html and if API related shown to the user, i was lead to believe eQSL stuff was but guessing that isn't the case so will go back and look

magicbug, 2020-05-26:
As I can't really action this any further the two items in question have their own issues and will be dealt with, please be aware it's noted and will be pushed for in future editions to the code and when accepting PRs

closed 2020-06-04 by magicbug