magicbug/Cloudlog #526: Clublog Upload failed

dg9vh, 2020-06-01:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
After last update of codebase upload to clublog failed via crontab and on shell:
huebelon@s074 [~/cloudlog]# curl https://cloudlog.dg9vh.de/index.php/clublog/upload/DG9VH
**SSL certificate problem: certificate has expired**
_Error Nothing awaiting upload to clublog for DG9VH/p
Nothing awaiting upload to clublog for DG9VH/p
Nothing awaiting upload to clublog for DG9VH/p_

The 3 messages on DG9VH/p are ok, there are no QSOs on queue, but for DG9VH there are some to upload but not working.

dg9vh, 2020-06-01:
Hi,

solved this issue (temporarly) by adding `curl_setopt($request, CURLOPT_SSL_VERIFYPEER, false);` in line 75. I think, this is an issue with my hoster, because after this, my queued QSOs where uploaded to clublog.

Maybe someone else also has this issue?

ok1cdj, 2020-06-02:
Also not worlking for me:
SSL certificate problem: certificate has expiredError

dg9vh, 2020-06-02:
Here an email from my hoster's-support:

> Hallo Kim
>
> Vielen Dank für Deine E-Mail.
>
> Ich habe mir das Zertifikat für clublog.org mal genauer angesehen. Die damit verbundene Zertifikat der COMODO RSA Certification Authority ist am 30.05. abgelaufen, was wohl das Problem > verursacht:
>
> Sat, 30 May 2020 10:48:38 UTC (expired 2 days, 21 hours ago) EXPIRED
>
> Du kannst das Resultat unter https://www.ssllabs.com/ssltest/analyze.html?d=clublog.org&s=2a02%3a16e0%3a1%3a33%3a0%3a0%3ac10b%3a109&latest einsehen.
>
> Ich hoffe, Dir mit diesen Informationen geholfen zu haben. Bei weiteren Fragen stehen wir Dir gerne zur Verfügung.
>
> Liebe Grüsse

In short: the corresponding CA-Certificate of COMODO RSA CA is expired and needs to be refreshed...

So we have two ways to get rid if this: 
1. See above, set curl-option...
2. Just make a cup of tea and wait...

As I made 1. with the opinion that we are not really transporting any information (beside password and so on) that needs to be secured. The thing is, transport would be secured as before, but we can't trust, that clublog is clublog :-) A thing, that I can accept for my personal situation.

on4dgt, 2020-06-03:
There was an security update on my Ubuntu-Cloudlog-server wich updates the ca-certificates.
I assume it updated the local corresponding CA-cert for COMODO and so solved the problem.  :-)

magicbug, 2020-06-03:
Looks like this isn't an Cloudlog issue but rather server one which seems to be addressed, prefer not to turn off SSL for everyone as I think it has its benefits.

closed 2020-06-03 by magicbug