magicbug/Cloudlog #535: Hawaii showing as worked status in dxcc list, but no Hawaii contacts.

mbridak, 2020-06-09:
Hawaii is showing as worked in the DXCC awards page, but I have no Hawaii contacts.

**To Reproduce**
Click Awards -> DXCC -> Show button

**Expected behaviour**
Not to show entities as worked/confirmed when no log exists for the entity.

**Screenshots**
Here's the shot showing worked.
[image]

Here's a shot after clicking the 'W' to show filtered log.
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Chrome
 - Version 83.0.4103.61 (Official Build) (64-bit)

dg9vh, 2020-06-09:
Hi,

Please update DXCC-List und Re-Check ALL after this...

[image]

Then retry again your DXCC-Awards list.

mbridak, 2020-06-09:
Did those.
[image]

[image]
[image]
[image]

But looks the same...

dg9vh, 2020-06-09:
vy strange... would be interessting, what your system doing there... @magicbug who did this? Maybe you could put him into this ticket as well?

mbridak, 2020-06-09:
This guy now shows as Hawaii in the log, but he's in Oregon....

[image]

And the WAS award shows no Hawaii

magicbug, 2020-06-09:
@AndreasK79

AndreasK79, 2020-06-11:
The only reason I can see that a mismatch between the award display and the qso display is happening, is that in the logbook-table, col_dxcc and col_country exists. If col_country is not set to the correct name according to col_dxcc, you will get this problem.

The award is based on col_dxcc, but the lookup is using col_country. The reason behind this, was that this was how the lookup was built before, and I re-used that.

mbridak, 2020-06-11:
I preface this with the statement that I have Zero codeigniter/query builder experience.

But in Logbook_model.php public function check_missing_dxcc_id($all). It calls check_dxcc_table($row['COL_CALL'], $qso_date).
And in check_dxcc_table() it just queries dxcc_prefixes with out first checking if there is anything matching in the dxcc_exceptions table.

I believe that's where the problem lies.

mbridak, 2020-06-13:
I submitted a pull request.

magicbug, 2020-06-13:
Thanks for the PR 👍

closed 2020-06-13 by mbridak