magicbug/Cloudlog #539: QRZ Upload Fails

myzinsky, 2020-06-17:
[image]

**Describe the bug**
QSOs of one Station Profile are not uploaded. 
**Additional context**
I'm using the current master branch.

AndreasK79, 2020-06-26:
I'm not sure what is going on here, but my guess would be that QRZ returns an error. There is no easy way to see that error right now. I have it on my TODO-list to present the errors for the user.

If you are comfortable in editing the code and run var_dump on a variable, the error can be seen in devtools in the browser.

AndreasK79, 2020-08-08:
#571  added now. Errors are logged in logfile.

closed 2020-08-09 by magicbug