magicbug/Cloudlog #540: Error when uploading ADIF

ofadam, 2020-06-18:
When uploading an ADIF file exported from Log4OM, I get the following error message. I've previously been able to upload my logs using the same workflow.

An uncaught Exception was encountered

Type: Error

Message: Call to undefined function mb_convert_encoding()

Filename: /var/www/ofadam.com/html/Cloudlog/application/libraries/Adif_parser.php

Line Number: 121

magicbug, 2020-06-18:
Care to share the adif file?

ofadam, 2020-06-18:
Aha! I figured out that I somehow didn't have Multibyte String installed for PHP... not sure how I overlooked that the past day. Sorry for the intrusion!

php-mbstring

closed 2020-06-18 by ofadam

k7ilo, 2022-01-22:
PERFECT....

Glad I looked this up.  It fixed my issue.
Thanks for the reply a long time ago.  LOL