magicbug/Cloudlog #544: How to backup adif via cron?

dg9vh, 2020-06-20:
I have no idea how to call the backup-routine via cron-job. Does anybody has this feature running?

AndreasK79, 2020-06-26:
I guess creating a cron-job with like this would work: curl --silent https://<URL-To-Cloudlog>/index.php/backup/adif/> &>/dev/null

If you visit that url your browser, it creates an adif from your log and places it in the backup folder.

dg9vh, 2020-06-26:
huebelon@s074 [~/cloudlog/backup]# curl  https://cloudlog.dg9vh.de/index.php/backup/adif/
huebelon@s074 [~/cloudlog/backup]# ls
./  ../  notes.xml

So nothing happens...

in browser it also does not work if not logged in... it only works in logged-in-state.

AndreasK79, 2020-06-26:
@dg9vh Ah yes, you are right, i didn't try that. @magicbug Are we missing something here Peter?

magicbug, 2020-08-10:
Fixed issue where it was wanting you logged in,

However if you run /index.php/backup/adif it will create an ADIF and store it

closed 2020-08-10 by magicbug