magicbug/Cloudlog #545: Need an option to export only QSOs not marked as sent to LoTW

nr0q, 2020-06-22:
When exporting now I can select a day or group of days to export. But uploading those produces duplicate QSOs in LoTW when trying to upload only those that haven't been already uploaded. I have some several hundred QSOs that get automatically uploaded to LoTW and eQSl by Gridtracker at the same that GridTracker sends the QSO to CloudLog. I later that day make a dozen SSB QSOs and log them directly into CloudLog. I'd like to export only those and mark them as QSO Requested thinking there might be a mechanism to export only those. Instead I had to export 400 QSOs and reupload all of the FT8 QSOs along with the handful of SSB ones.

AndreasK79, 2020-06-23:
Added #546

magicbug, 2020-06-27:
Merged and fixed

closed 2020-06-27 by magicbug