magicbug/Cloudlog #548: Frequency not correctly transmitted by API

myzinsky, 2020-06-24:
When sending a QSO with a specific frequency (eg. 14.2) is not correctly filled into the Cloud Log table using the API as can be seen in this picture: 

[image]

When I use 14.200000 instead cloud log will show it correctly. IIRC the number of zeros is not a relevant constraint for ADIF log files. Just a double in MHz...

magicbug, 2020-06-25:
Frequency is expected in Hz 14000000 as per adif spec. Same was what's expected on the Cloudlog Freq inputs

myzinsky, 2020-06-25:
[image]

But ADIF wants to have it not in Hz it wants the frequency in MHz. If I send 14.2 in the ADIF over the Cloudlog API it will turn into 1420 Hz. If I send 14.200000 with tail zeros it will turn into 14000000 correctly. The *Number* specification in the ADIF format does not mention that tail zeros are required down to the Hz level. 

Not sure if this is really *Not a bug*

magicbug, 2020-06-25:
Not its a bug and this has been the way I've handled freqs within cloudlog from the start, it will except a few varieties if you check the import system but largely its the easiest way to handle frequencies without having to put 50+ checks in

myzinsky, 2020-06-25:
Independent from the API, the same issue happens when I import an ADIF File. I have attached it to this post. When the Frequency is e.g. 14.1 Cloudlog shows 1420 Hz. In my opinion Cloudlog is therefore not ADIF compatible. Or do I get something totally wrong? 

Why not taking the ADIF value as double and multiply it with 10^6 and we have the frequency in Hz?

[adif.txt](https://github.com/magicbug/Cloudlog/files/4833942/adif.txt)

Yesterday I imported a huge amount of QSOs which where then forwarded in a wrong way to QRZ.com by Cloudlog.

myzinsky, 2020-06-28:
It is definitely a bug. I just imported an ADIF rendered by the well known Contest Software **UcxLog**. I have to manually change all the frequencies of hundreds of QSOs. :-(

myzinsky, 2020-06-28:
I checked the cloud log code for the ADIF import:

https://github.com/magicbug/Cloudlog/blob/36621b1e23174d93055e59fccf36847a4b48f4ef/application/models/Logbook_model.php#L1417

```php
        // Store Freq
        // Check if 'freq' is defined in the import?
        if (isset($record['freq'])){
            $cleanfreq = preg_replace('#\W#', '', $record['freq']);
            $freqlng = strlen($cleanfreq);
        }else{
            $freqlng = 0;
        }

        if(isset($record['freq']) && $freqlng < 7 ) {
            $cleansedstring = preg_replace('#\W#', '', $record['freq']);
            $freq = $cleansedstring."0";
        } elseif($freqlng >= 7) {
            $cleansedstring = preg_replace('#\W#', '', $record['freq']);
            $freq = $cleansedstring;
        } else {
            $freq = "0";
        }
```

With the ADIF spec in mind, this code makes no sense to me. Also I don't get why the length of 7 makes a difference.

I think something like this would make sense (pseudo code not tested):

```php
   if (isset($record['freq'])){ // record[freq] in MHz
            $freq = floatval($record['freq']) * 1000 * 1000; // store in Hz
        }else{
            $freq = 0;
        }
```

m1geo, 2020-09-04:
Just spent half an hour, poking at the same bug, having dismissed this issue as I was supplying a "full" frequency (say, "21.29900", as LoTW exports).

I have some comments on #605 which I have closed as duplicate of this. Most of the rambling is wrong, but I concluded the frequency in the database is stored wrongly.

m1geo, 2020-09-04:
I'm just about to run a fresh import of 50k contacts with the following:

_Check for Freq_
```
if (isset($record['freq'])){ // record[freq] in MHz
    $freq = floatval($record['freq']) * 1000 * 1000; // store in Hz
} else {
    $freq = 0;
}
```

and

_Check for RX Freq_
```
if (isset($record['freq_rx'])){ // record[freq_rx] in MHz
    $freqRX = floatval($record['freq_rx']) * 1000 * 1000; // store in Hz
} else {
    $freqRX = NULL;
}
```

I'll report back with how it goes. There's a good mix of VLF (136 kHz stuff) and microwave (24 GHz) stuff, as well as the more typical contacts...

myzinsky, 2020-09-04:
Haha, great to see that you just used my pseudo code :P I'm keen to know if that works with so many contacts!

m1geo, 2020-09-04:
Well it totally failed. All QSOs imported with no frequency :) Time to debug!

myzinsky, 2020-09-04:
Maybe the 1000.0 must be inside the parenthes

m1geo, 2020-09-04:
Nope, the code at https://github.com/magicbug/Cloudlog/issues/548#issuecomment-687118601 works fine. I was just blind! 👍  All 50k contacts have the correct frequency in them.

m1geo, 2020-09-04:
@myzinsky I duplicated your code for receive and make PR https://github.com/magicbug/Cloudlog/pull/607.

magicbug, 2020-09-05:
Can everyone confirm this is OK and that I should merge it?

closed 2020-09-05 by magicbug

m1geo, 2020-09-12:
Sorry Pete. Just see this. I have imported around 50000 QSOs with it, and it worked fine. That's all the testing I could do. But it's in essence @myzinsky mod tidied a little.