magicbug/Cloudlog #551: Save QSO Button - nothing happens

5foot8, 2020-06-27:
**_Do you need to sign up for all the online logging websites and QSL websites before CloudLog become fully functional?_**
I'm attempting to just log QSO's as a short-wave listener but after adding callsign and linking to my station and the frequency  from the CAT link **when I press the Save QSO it does nothing, it's like the button is non functional.**

I do expect to gain my foundation license in the next month so can sign up for all the sites that require a callsign, but as a SWL I'm stuck, and I don't know if not having LoTW and QRZ.com linked is causing the trouble.

Cloudlog is installed on a local Raspberry Pi and it appears to be fully working after around seven attempts to install and add all the privileges to SQL database and other trip hazards.

I don't mind waiting until I have a callsign if that is what is breaking the logging software, but if something is wrong I'd like to get it fixed ready for when I can actively participate in the wonderful learning experience.  I have turned off all the security features I can find in the browser but still no joy.

magicbug, 2020-06-27:
For no QSO to be added it could be a few things

- no updated country files
- No station profile

5foot8, 2020-06-27:
> For no QSO to be added it could be a few things
> 
> 
> 
> - no updated country files
> 
> - No station profile

Thank you - a full country file update has brought the button to life!  Now fingers crossed for the 12th!

closed 2020-07-01 by magicbug