magicbug/Cloudlog #552: Consolidated Statistics over all Station Profiles

myzinsky, 2020-06-28:
Having already 8 Station Profiles for different SOTA, COTA and Portable Activities, I discovered that the statistics are only shown for the active logbook. However, for some awards a consolidated view would be desired. Therefore, I recommend a checkbox in the settings to enable a consolidated view over all station profiles.

ryandzink, 2020-07-06:
I would add that this would be very useful for people who are roving for satellite passes. I've had to create new station profiles for new grid squares when I'm traveling, and the grid square analytics for only my home station aren't as useful when I know there are other grid squares I have only contacted while roving.

myzinsky, 2020-09-03:
I did a quick check. Mostly the ```WHERE station_id = ...``` sql statements should be included in the queries or not, depending on a configuration switch.

@magicbug would that be okay? I could work on some PRs.

magicbug, 2020-09-03:
I'll try to write a reply, apologies if it's jumbled, lots of pain and thoughts are a bit over the shot.

Yes `WHERE station_id =` is used to get the data for that QSO profile, I'm not sure the best way to group, I did sit wondering this week during LoTW dev whether adding a "Logbook" to sit in front of Station Profiles might be an idea for example

- Create a Logbook (Basically just a text field and an ID
- Then Station Profiles are assigned to the Logbook (One to Many style database relation)

This would then make doing grouped stats easy as these areas would then be built of the Logbook associations if that makes sense.

I'll let @dg9vh @AndreasK79 chime in they seem to know my thinking well of late 👍

AndreasK79, 2020-09-04:
@myzinsky Making a config-approach was my first thought too, when this issue was brought up a few months back. I made a quick test, and it's not difficult to do. The problem arises when you have several logs, and they are not from the same DXCC (or other criterias I can't think of right now).

@magicbug Some form of grouping is necessary, and your approach seems to be a logical way to accomplish this. A GUI is needed for this, and I guess the active logbook-group should be found from the active profile.

myzinsky, 2020-09-04:
@magicbug just to get you right: There would be a structure like this?

[image]

The charming thing about this is, that even if there are multiple users with multiple logins, one can create one log per user, and they could have all their own statistics. :-) So I would not have to maintain two cloud log instances one for myself and one for my father.

magicbug, 2020-09-05:
@myzinsky Yes exactly that

User can have many logbooks, logbooks can have many Station Profiles

This would also work well for upcoming needed Auth changes, so user_id would be assigned to the logbook.

myzinsky, 2020-09-05:
Yes, after a short discussion with @dg9vh we also thought that we could have the same station profile in many different logbooks. But that would also work.

magicbug, 2020-09-05:
Created a new issue for Creation of Logbooks. https://github.com/magicbug/Cloudlog/issues/609

closed 2020-12-11 by magicbug