magicbug/Cloudlog #556: Error LOTW Import

myzinsky, 2020-07-05:
When pulling LOTW Data I get recently this error:

[image]

dg9vh, 2020-07-05:
This sometimes happens also to me on days selected for starting and having absoultely no confirmations on lotw-side on it...

Please check date starting your import... maybe you selected a date in future?

tekk, 2020-07-07:
Same issue here.

kd8skz, 2020-07-12:
Go ahead and add me to this list as well, I thought it was a issue with my database, but I guess not.

magicbug, 2020-08-10:
I've only see this happen when LOTW services are not working, the feature certainly seems to be working.

I've added some further error checking to make this more graceful

closed 2020-08-10 by magicbug