magicbug/Cloudlog #557: Callsigns with / show up as new contacts (highlighted red) even though they correctly show previous QSOs

ryandzink, 2020-07-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'QSO -> Post QSO'
2. Enter details for a QSO with a station that has a / in it (in my case K8YSE/7)
3. Previous contacts with that station are correctly shown on the sidebar
4. The callsign field is highlighted red and indicates this is a new callsign

**Expected behaviour**
The callsign field should be green and contact details for the callsign should auto populate. I have only seen this happen on callsigns with slashes in them so I am guessing it is something to do with character escaping? This particular contact is happening for the same satellite/bands on which I have made contact in the past
[image]
.

**Screenshots**
1 screenshot attached

ryandzink, 2020-07-07:
I don't think this is isolated to callsigns with slashes since I just experienced the same with a callsign with no slashes. Am I just not understanding the criteria that would cause the callsign to be considered a new callsign?
[image]

magicbug, 2020-08-10:
With the above screenshot with FM/2M is the Satellite tab set? if not then based on the screenshot that is a new callsign for the band/mode combination and how it's designed to work.

ryandzink, 2020-08-11:
@magicbug maybe I am misunderstanding what criteria cloudlog is taking into account for considering something a new callsign. For satellites is it that you have never worked that contact on that satellite?

I just tested this out with K8YSE/7 above, who I have made contact with on AO-27 in the past, and even after choosing that satellite on the Satellite tab he still shows up as a new callsign with a red border, even with that previous AO-27 contact displaying under the map.

magicbug, 2020-09-10:
If that's happening then there's a bug

If I work K8YSE/7, the next time I work John it should just mark it as Green once I tab out the callsign field.

commit b8d2032e (2021-01-29): [QSO] Fixes #557 Where /P wasn't handled properly
closed 2021-01-29 by magicbug