magicbug/Cloudlog #563: Ability to do tasks in batches

M0TZO, 2020-07-24:
I had an issue (maybe due to host, but even on their faster MySQL that larger ADIF files would time out.

I was thinking that for this and country files (although this one hasn't timed out, yet) to do in batches.

I can upload in theory up to 512Mb (largest I've dealt with was 8Mb), but times out after processing about 1,000-1,500 QSOs.

If I could upload a whole ADIF, then say 1,000 and come back later and do next batch when importing, could be useful.

Hope this makes sense.
_Yes, maybe need new host_.

dg9vh, 2020-07-24:
Hi, if you have access to your php-configuration, you can take a try with the max_excecution_time-parameter and set it up to some minutes instead of seconds... default is 30 seconds if I remember correctly. Could be too short for large files.

magicbug, 2020-08-09:
Does make sense but currently i've no plan to split ADIF files on upload, this really is a host limitation

magicbug, 2020-08-09:
Added as a feature item

M0TZO, 2020-08-09:
I've set mine to 300 secs now (was 30) in the php settings for host. But some functions I don't need to do all logs like the update country files Re-Check.

closed 2020-12-11 by magicbug