magicbug/Cloudlog #565: Parse error, syntax error in Award site

torsten-ernst, 2020-07-26:
Hi,

When I click on Award I get the following error messages. Any ideas?

```
Parse error: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING) in /www/htdocs/v035197/Cloudlog-master/application/models/Dxcc.php on line 171
A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /www/htdocs/v035197/Cloudlog-master/application/models/Dxcc.php:171)

Filename: core/Common.php

Line Number: 570

A PHP Error was encountered
Severity: Parsing Error

Message: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING)

Filename: models/Dxcc.php

Line Number: 171
```

73 de Torsten - DL9GTB

jtmnt, 2020-07-26:
Make sure you are running php 7. 

See also #356

torsten-ernst, 2020-07-26:
Thanks for the hint.

73 de Torsten - DL9GTB

closed 2020-08-09 by magicbug