magicbug/Cloudlog #566: Allow for updating LotW/eQSL status in the QSL edit context menu

ryandzink, 2020-07-26:
**Is your feature request related to a problem? Please describe.**
When updating the status of QSLs in LotW, it takes some time to go into the edit QSL page and mark the QSL as received when the other station confirms contact (several clicks and page loads). When you have a large number of QSLs to go through this becomes a very repetitive process.

**Describe the solution you'd like**
Is it possible to add either eQSL or LOTW "mark QSL received" actions to the context menu in the log? They could be contextually added like the columns for their status are added to the log (i.e. only if the user has configured a username for each service). I can't speak for others but the number of physical QSL cards I go through is far fewer than the LOTW confirmations I do, so the current context menu only supporting QSL cards is rarely used.

magicbug, 2020-09-05:
I'd generally suggest letting Cloudlog grab data direct from LoTW or eQSL rather than having it as a box to mark as received, and as there already a function todo this in edit I don't see any advantage to adding it to the drop down as these are electronic.

ryandzink, 2020-09-08:
@magicbug I think you're right, especially with all the new LotW syncing features. I'm going to close this out.

closed 2020-09-08 by ryandzink