magicbug/Cloudlog #573: Databasename with dots not working

do5cth, 2020-08-11:
Hi,

i have my Databasa-Name with dots "cloudlog.do5cth.de".
Cloudlog cant run with these after Install.

Following Error on site:

A Database Error Occurred
Error Number: 1064

You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '.`do5cth`.`de`' at line 1

SHOW TABLES FROM `cloudlog`.`do5cth`.`de`

Filename: libraries/Migration.php

Line Number: 167

AndreasK79, 2020-08-29:
They are aware of the issue, but it seems not a priority to fix: https://github.com/bcit-ci/CodeIgniter/issues/3719

magicbug, 2020-09-04:
Sadly this is a limitation of the framework and isn't likely to get resolved. 

Thus closing

closed 2020-09-04 by magicbug