magicbug/Cloudlog #574: Unable to import from LOTW

therokh, 2020-08-15:
**Describe the bug**
Error when trying to import LOTW

**To Reproduce**
Steps to reproduce the behaviour:
1. Fresh install
2. Set up a new admin user and login
3. Set up profile including LOTW username and password
4. Update country list
5. Try to import LOTW and click Import

**Additional context**
It seems #324 had this problem, but I have confirmed my LoTW login details work correctly.

`A PHP Error was encountered
Severity: Notice
Message: Undefined variable: lotw_table_headers

Filename: lotw/analysis.php
Line Number: 4
A PHP Error was encountered
Severity: Notice
Message: Undefined variable: lotw_table
Filename: lotw/analysis.php
Line Number: 5
`

dg9vh, 2020-08-15:
Did you create a station profile?

therokh, 2020-08-15:
Yes and set it as active

jfedyczak, 2020-08-15:
+1 doesn't work for me either

magicbug, 2020-08-15:
Not seeing a problem here so this is localised issues.

- Make sure logins are right
- make sure you have the PHP curl module.
- Share outputs of phpinfo();
- check Apache/PHP server logs for anything behind the scenes that's failing.

jfedyczak, 2020-08-15:
Error reported is exactly as therokh posted above. PHP info output can be seen [here](https://www.dropbox.com/s/huke7gyp4t5r1jy/phpinfo.html?dl=0). Curl is present.

therokh, 2020-08-16:
On a hunch I changed my LOTW password to one without symbols, updated my Cloudlog profile, and then tried import again.

It changed to repeating this error for a minute or so as it appeared to import:

```
A PHP Error was encountered
Severity: Notice
Message: Undefined variable: state
Filename: controllers/Lotw.php
Line Number: 96
```

Then I went back to the home page and my QSOs were there.

magicbug, 2020-09-06:
Made quite a few changes to the LoTW code this should hopefully now all be addressed

closed 2020-09-06 by magicbug