magicbug/Cloudlog #582: Days with QSO new feature

M0TZO, 2020-08-24:
I started this year with the target of getting 1 QSO for each day (366 contacts). Thanks to Covid I smashed that.

On the Days with QSO's I think a nice addition would be Longest Streak (_most days with at least 1 QSO in a row_), or even the top few periods. I am trying for daily QSO's but not sure how long I've done it for without looking through pages of this years logbook.

magicbug, 2020-08-24:
I'm guessing your seeing this as an addon to /index.php/dayswithqso ?

M0TZO, 2020-08-25:
Yes, seems the most logical place, or the dashboard (_Important  You have made no QSOs today, time to turn on the radio! / You have had 1 QSOs Today!_) bar

AndreasK79, 2020-08-26:
This was an interesting one. I've never done any type of query like that. After a lot of searching, I managed to find one that works with MySQL:

> WITH qso_dates AS (
	 SELECT DISTINCT cast(col_time_on as date) qso_date
	 FROM table_hrd_contacts_v01 thcv 
	 WHERE station_id = 1
),
qso_date_groups AS (
   SELECT
	   qso_date,
	   qso_date - CAST(row_number() OVER (ORDER BY qso_date) as INT) AS grp
	 FROM qso_dates
   )
SELECT
   max(qso_date) - min(qso_date) +1 AS streak, max(qso_date), min(qso_date)
FROM qso_date_groups
GROUP BY grp
ORDER BY streak DESC
LIMIT 10

The query above lists the 10 longest streaks. @M0TZO If you are able to run queries, and know your station_id, it would be interesting to see if this query agrees with your spreadsheet.

@magicbug What are the requirements for Cloudlog regarding MySQL version? I read that versions before 8 doesn't support the WITH-function.

M0TZO, 2020-08-26:
Sorry, I don't know enough about testing the above code.

M0TZO, 2020-08-26:
Figured it out, but I'm not sure what version, but I guess it's not 8 (yet)
```
Error
Static analysis:

1 errors were found during analysis.

Unrecognized statement type. (near "WITH" at position 0)
SQL query: Documentation

WITH qso_dates AS ( SELECT DISTINCT cast(col_time_on as date) qso_date FROM table_hrd_contacts_v01 thcv WHERE station_id = 1 ), qso_date_groups AS ( SELECT qso_date, qso_date - CAST(row_number() OVER (ORDER BY qso_date) as INT) AS grp FROM qso_dates ) SELECT max(qso_date) - min(qso_date) +1 AS streak, max(qso_date), min(qso_date) FROM qso_date_groups GROUP BY grp ORDER BY streak DESC LIMIT 10

MySQL said: Documentation

#1064 - You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'qso_dates AS (
SELECT DISTINCT cast(col_time_on as date) qso_date
FROM table_h' at line 1
```

magicbug, 2020-08-26:
@AndreasK79 We dont have a min mysql-server/mariadb version, I will be the first to admit I still have mysql-server 5.7 in production!

AndreasK79, 2020-08-28:
The query above is not good. I wrote my own php code to find max streaks. When comparing the query against my code, I found that it didn't work as it should.

While researching this problem, I found several queries that were supposed to give me max streaks. One of them gave me the same number as my own code, but without dates. I then checked the results, and I found it to be correct.

I'll see if I can make a PR which adds this to the "Days with QSO" page.

AndreasK79, 2020-08-29:
Pushed #584

M0TZO, 2020-08-29:
Looks good. Only thing is not showing current days with QSOs, which should be at 47 but not listed at all.

AndreasK79, 2020-08-29:
@M0TZO Do you mean your max current streak, or do you mean that it should have been shown in the table? I chose to just show the 10 highest streaks. Perhaps a note about that should be added.

magicbug, 2020-08-29:
Seems I had a brain fart when I merged this code, does it still require the latest MySQL? if so it will need some check placed to make it only available on it

Error Number: 3065

Expression #1 of ORDER BY clause is not in SELECT list, references column 'logbook_2m0sql_master.thcv.COL_TIME_ON' which is not in SELECT list; this is incompatible with DISTINCT

select distinct cast(col_time_on as date) as date from TABLE_HRD_CONTACTS_V01 thcv where station_id = 1 order by col_time_on asc

Filename: models/Dayswithqso_model.php

Line Number: 87

magicbug, 2020-08-29:
Made commit https://github.com/magicbug/Cloudlog/commit/30f0b6b3e7ca4aa6633a3b9d2fdef878ada427b2 to see if it fixes the sql error

magicbug, 2020-08-29:
I've reverted the code push as it was a breaking change due to CAST being MySQL 8.0 only.

AndreasK79, 2020-08-29:
@magicbug Are you sure that cast is a MySQL 8.0 function? What if you change to order by date, as that is an alias given to the cast column.

AndreasK79, 2020-08-29:
@magicbug Cast is in 5.7: https://dev.mysql.com/doc/refman/5.7/en/cast-functions.html, so that is not the reason for this error.

AndreasK79, 2020-08-29:
@magicbug Can you test this query then? (Substitute tablename and station_id).

> select distinct cast(col_time_on as date) as date from table_hrd_contacts_v01 where station_id = 1 order by date asc

AndreasK79, 2020-09-09:
@magicbug Could  you test the query above and see if it works or not?

M0TZO, 2020-09-09:
When edited I got the following:-

2006-11-22
2006-11-24
2006-11-25
2006-11-26
2006-11-27
2006-11-28
2006-11-29
2006-12-02
2006-12-05
2006-12-09
2006-12-11
2006-12-12
2006-12-16
2006-12-17
2006-12-20
2006-12-23
2006-12-24
2006-12-25
2006-12-26
2006-12-27
2006-12-28
2007-01-02
2007-01-08
2007-01-14
2007-01-16

M0TZO, 2020-09-09:
Oh, list was longer, this was 1st 25.

AndreasK79, 2020-09-09:
@M0TZO Thanks Paul. It should work, but since Peter got an error on his server, it would be nice to know if it works there.

magicbug, 2020-09-10:
> select distinct cast(col_time_on as date) as date from table_hrd_contacts_v01 where station_id = 1 order by date asc

Sorry meant to check this yesterday but rough 24 hours, yes works perfect

AndreasK79, 2020-09-10:
@magicbug Thanks Peter. I'll submit a new PR.

@M0TZO Paul, I never got an answer to my question above...
Did you mean your max current streak, or did you mean that your current streak should have been shown in the table? I chose to just show the 10 highest streaks. It can be extended to include a current streak as well, just need to figure out the code for it.

AndreasK79, 2020-09-10:
#612 submitted. This can be extended, but one step at a time.

M0TZO, 2020-09-13:
I wanted the current one too. Which should be 2nd on my table according the what's showing. Nice work on it.

AndreasK79, 2020-09-13:
@M0TZO I'll see if I can add the current streak as well. Does the numbers look correct to you?

M0TZO, 2020-09-13:
Seems to be. I've not actively checked, but seems similar to before and another Cloudlog install looks right too.

AndreasK79, 2020-09-13:
Thanks. It did seem right to me when writing this. I checked my longest one, and the count was correct. I'll have a go at the longest current streak as well.

magicbug, 2020-09-15:
Just to say I've change the date format on the Days with QSOs to follow the users date format selection

AndreasK79, 2020-09-21:
#625 added. @M0TZO here you go :)

AndreasK79, 2020-09-22:
If no objections, I guess this can be closed now.

magicbug, 2020-09-22:
Agreed

commit 30f0b6b3 (2020-08-29): dayswithqsos added "col_time_on" to "getDates" to see if it fixes error

Change to see if it fixes #582
closed 2020-09-22 by magicbug