magicbug/Cloudlog #583: API Sync in Both Directions

myzinsky, 2020-08-27:
Hi Peter, 

do you have an idea how we could enable synchronisation in both ways? Right now I can synchronise from my app only into cloudlog. But keeping both things automatically in sync would be super awesome!

Best,
Matthias

magicbug, 2020-08-28:
Guessing you want to be able to call a json end point to get last x qsos?

myzinsky, 2020-08-28:
I think I'm pretty flexible here, JSON would be a natural choice. I think it would be great if we could find a way to sync the QSOs in a way that basically the database of my app and also the database of could log could be in sync. Probably we would need to sync also an unique ID/hash because if a QSO is changed we cannot just compare the QSOs by the values in the fields. 

Do you have any idea here?

dg9vh, 2020-08-28:
Hi, maybe using the COL_PRIMARY_KEY within the TABLE_HRD_CONTACTS_V01 could help identifying the QSO to update? If this value would be exported to Matthias' app, he could refer to this key when updating rows.

myzinsky, 2020-08-28:
I think one problem arises then, what if in the app and in the cloudlog itself QSOs are added, that have suddenly the same key :)

dg9vh, 2020-08-29:
Using key+prefix combination

magicbug, 2020-08-29:
**COL_PRIMARY_KEY** within the  TABLE_HRD_CONTACTS_V01 seems to be the best way of identifying a QSO, this is generated once a QSOs saved, which is fine, so my question was is whats being requested here a json export of x number of recent QSOs?

myzinsky, 2020-08-30:
Okay so the sync algorithm would then be in the following way:

1. Upload QSOs from CloudLogOffline to Cloudlog which are not yet marked as "Synced" or are marked as "Updated" over current API
2. For each Upload Cloudlog returns the COL_PRIMARY_KEY id in the returned JSON, for example:
    ```json
    {
       "status":"created",
       "key":"123",
       "type":"adif",
       "string":"<call:5>...<eor>"
    }
    ```
3. CloudLogOffline annotates after successful sync this ID in the internal app sql table --> QSO is marked as synced.
4. CloudLogOffline requests ALL QSOs from CloudLog which are provided as JSON, the QSO format shall be ADIF, as usual.

5. If a QSO is changed in CloudLogOffline it must be marked as "Updated" and we need an update mechanism in the API for existing keys.

Thoughts? I don't like the fact that ALL QSOs must be downloaded, but it would be the easiest way?

myzinsky, 2020-09-12:
Maybe a timestamp for last change per row? Thoughts?

myzinsky, 2020-10-18:
@magicbug any inputs from your side here? I think in the app I will have two views later on. The *Archive* which will be synced with Cloudlog. And the *Desktop* (any other catchy name? recommendations welcome) where the actual SOTA activity or contest will be logged locally. Then it could by synced to cloudlog and then it pops up also in the *Archive*. I think I will wait until this station-profile stuff is sorted out, nevertheless maybe we could already agree on a strategy here.

closed 2020-12-11 by magicbug