magicbug/Cloudlog #588: Notification of LoTW status when entering a callsign is inconsistent

Ciemon, 2020-08-29:
After entering a station's callsign their DXCC country is indicated below, LoTW is also indicated but that indication is inconsistent. Whilst 2M0SQL gets the LoTW label G0ABI and DL6KBG do not despite the fact that I've had previous LoTW confirmations from them. 

When entering a PostQSO
1. Set the satellite
2. Set station details
3. Set the correct time on the QSO page
4. Tab into the callsign field and enter G0ABI. 
5. Tab into the next field and the England label appears below G0ABI.
 
It's expected that in addition to England, LoTW should appear under the callsign field.

Ubuntu Linux 20.04
Chromium version 84.0.4147.135 (Official Build) snap (64-bit)

magicbug, 2020-08-29:
Looks like its not a bug per say, but script doesn't always update. however I really want to change how this works so rather than imports into the database it just checks the text file to try reduce having to process it into SQL as it takes for ages!

magicbug, 2020-08-29:
I shall add this to my LoTW todo list as I'm in this area of code currently with the sync stuff

magicbug, 2020-09-21:
Just to say that this has now been implemented, the csv is downloaded and then PHP queries it directly to find if a callsign is in the file. no cronjobs need changing, I've included the file in git and PHP is set to update it once a week with the current recommendations.

I'll add a migration to delete the lotw_user table next

magicbug, 2020-09-21:
And that is now done!

closed 2020-09-21 by magicbug