magicbug/Cloudlog #593: Creating new Station returns SQL error

matthewrwright, 2020-09-01:
Environment: Self hosted clean build Ubuntu 20 Docker container, PHP7.4 

**Describe the bug**
When creating a new station the field "Station DXCC" is a label not a field causing empty string to be posted in SQL Query

A Database Error Occurred
Error Number: 1366

Incorrect integer value: '' for column 'station_dxcc' at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`, `state`, `eqslqthnickname`, `qrzapikey`) VALUES ('Home QTH', 'IOxxxx', 'xxxxxxxxxxx', '', '', 'N0CALL', '', '', '', '14', '27', '', '', '')

Filename: models/Stations.php

Line Number: 52

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Admin > Station Profiles > Create '
2. Submit a new station
4. See error

**Expected behaviour**
The form field should be visible and entered into the database on submit.

**Screenshots**
<Not required>

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.01
 - Browser Chrome
 - Version 84.0.4147.105 (Official Build) (64-bit)
(also confirmed Firefox 84.0)

**Additional context**
Clean install, empty database basic configuration performed.

matthewrwright, 2020-09-01:
**Partial Resolution** Install php-xml package and Admin > Update Country Files resolved issue - suggest possibly making this either a warning on the Station Profiles > Create page or loading an initial list as part of the Install routine?

magicbug, 2020-09-01:
Should be a warning on login, but it's also listed in the wiki install info.

closed 2020-09-01 by magicbug