magicbug/Cloudlog #599: Database Error after last git pull when selecting LotW Menu Entry

jpaechnatz, 2020-09-03:
**Describe the bug**

After the last git pull this morning, LotW does not work any more. More precisely, the Menu changed from LotW Import to LotW.

**To Reproduce**

Selecting the LotW Menu Entry causes the following error:

A Database Error Occurred
Error Number: 1055

Expression #1 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'cloudlog.lotw_certs.lotw_cert_id' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by

SELECT * FROM `lotw_certs` WHERE `user_id` = '6' GROUP BY `callsign` ORDER BY `cert_dxcc` ASC

Filename: models/LotwCert.php

Line Number: 23

Ciemon, 2020-09-03:
I too have this same error this morning, 3rd Sept 20.

magicbug, 2020-09-03:
I know @Ciemon uses my company magicbug for hosting so I can fix that, this is due to MySQL having ONLY_FULL_GROUP_BY turned on, I personally tend to keep this off.

mysql > SET GLOBAL sql_mode=(SELECT REPLACE(@@sql_mode,'ONLY_FULL_GROUP_BY',''));

jpaechnatz, 2020-09-03:
Since cloudlog is my only db in mysql (other stuff is using postgres ;-)) above command applied, that solved it. Thx.

vy 73 de DB2PJ ;-)

closed 2020-09-03 by magicbug