magicbug/Cloudlog #600: Error when clicking on LOTW export

myzinsky, 2020-09-03:
[image]

I assume that is something because you did some commits for LOTW?

myzinsky, 2020-09-03:
Negative, the table and fields are there. Seems to be something with the servers settings...

myzinsky, 2020-09-03:
Adding this to mysql config solves the problem:
```
sql_mode = "STRICT_TRANS_TABLES,NO_ZERO_IN_DATE,NO_ZERO_DATE,ERROR_FOR_DIVISION_BY_ZERO,NO_AUTO_CREATE_USER,NO_ENGINE_SUBSTITUTION"
```

magicbug, 2020-09-03:
Made a commit that will mean you no longer need todo that change as it seems quite a few are running MySQL in strict mode

closed 2020-09-03 by magicbug