magicbug/Cloudlog #604: Improve ADIF Import time by batch inserting

magicbug, 2020-09-04:
The code behind ADIF importing is fairly old and needs some live basic idea is

- instead of single inserts build into one batch hit.
- double check all verification checks on QSOs are happening as fast and simple as possible.

m1geo, 2020-09-04:
Can't help feel that I inspired this one ;) Thanks Pete!

magicbug, 2020-09-05:
@m1geo you was but needs doing

AndreasK79, 2020-11-30:
#725 was pushed now with modifications to the adif_parser. On my local machine, this made a BIG difference.

AndreasK79, 2020-11-30:
I see room for further improvements by changing the logic on inserting of QSOs.

1. For each QSO, a call to the database is done to check for QRZ API Key.
2. For each QSO, a call to the database is done to check for sub mode.
3. For each QSO, a call to the database is done to get the country from the DXCC-table.
4. For each QSO, if Cloudlog is trying to determine the DXCC, several calls are made to the database trying to figure out the DXCC. This is by design, so not easy to get around, unless you trust the DXCC that are in the ADIF file, and check "
Use DXCC information from ADIF".

closed 2020-12-11 by magicbug