magicbug/Cloudlog #609: Creation of a "Logbook" to link Station Profiles together

magicbug, 2020-09-05:
Purposing that a new item "Logbook" is created which links to Station Profile, as in a Logbook can have MANY Station Profiles this will allow them to be grouped together for award purposes or for example you're doing portable but under the same callsign and many locations but want to see Stats/Awards.

**Required Items**

- New table called "Logbooks" with rough IDs of - id, user_id, logbook_name, logbook_description, created_date, updated_date
- Frontend to capture the creation of Logbooks, edit them and also view them.
- System in place for current users to get them to create logbooks and allocate station profiles to them.

**Optional Extras**

- Allow Logbooks to have multiple users, each user would have their own station profiles, this would need different database logic and an extra table.

magicbug, 2020-09-05:
Other considerations that need to heavily be considered is all database schema, everything needs to have primary keys, indexes, and all calls need to be tested and optimised to make them fast as possible.

I'm getting more rigid on this when it comes to code additions.

magicbug, 2020-09-05:
The other option for making things easier with multi-user, just make the logbook MANY to MANY with users then all users can access all station profiles as log as the Logbooks they have access to can see the station profile

m0pcb, 2020-09-05:
I think also has implications for LoTW and so on. So would still need to
pick parameters from station profile.

But yes, a Logbook that contains one or many station profiles is a great
idea. This would work well for individuals as well as club stations that
pass around call signs to different locations.

I wonder if there may be a need to limit a Logbook to only contain station
profiles that are the same DXCC/Country. That would mean that awards and
stats aggregation is valid across an entire logbook. Presumably, the DXCC
and location details would still be stored in the station profiles, but
conceivably some of it could move to the Logbook, but that would be extra
work.

I think that any given station profile can only be a member of one logbook.

Being able to assign existing station profiles to a logbook is essential.

Each award and stats pages would need to pull data based on logbook, and
ideally also be able to select which logbook, without having to make that
logbook the active one.

May have more thoughts on this idea, but hopefully useful first pass of
thoughts…

73, Iain AD5XI / M0PCB

*From:* Peter Goodhall <notifications@github.com>
*Sent:* 05 September 2020 18:16
*To:* magicbug/Cloudlog <Cloudlog@noreply.github.com>
*Cc:* Subscribed <subscribed@noreply.github.com>
*Subject:* [magicbug/Cloudlog] Creation of a "Logbook" to link Station
Profiles together (#609)

Purposing that a new item "Logbook" is created which links to Station
Profile, as in a Logbook can have MANY Station Profiles this will allow
them to be grouped together for award purposes or for example you're doing
portable but under the same callsign and many locations but want to see
Stats/Awards.

*Required Items*

   - New table called "Logbooks" with rough IDs of - id, user_id,
   logbook_name, logbook_description, created_date, updated_date
   - Frontend to capture the creation of Logbooks, edit them and also view
   them.
   - System in place for current users to get them to create logbooks and
   allocate station profiles to them.

*Optional Extras*

   - Allow Logbooks to have multiple users, each user would have their own
   station profiles, this would need different database logic and an extra
   table.

closed 2020-12-11 by magicbug