magicbug/Cloudlog #618: Edit QSO does not reflect 'Measurement Preference' parameter in Account

timkirby, 2020-09-16:
The Edit QSO feature does not seem to reflect 'Measurement Preference' parameter in the Account page.

My measurement preference is set to KM but Edit QSO distances are displayed in miles.

Thank you :-)

AndreasK79, 2020-09-16:
I tried to reproduce, but it works as intended here. When I go to user profile and change, it changes accordingly.
Does it change at all if you change in user profile?

timkirby, 2020-09-16:
Hi Andreas,

Thank you. Yes, I made a change to the profile (just changed the date format) and then the measurement change was applied and I’m now in KM!

Tim

AndreasK79, 2020-09-16:
Hi Tim.

That's good to hear. It's difficult to say what the problem is, but the first time you run this code, no value exists in the database for measurement. It picks up the value from the config file as a fallback value. Do you know what value is set in the config file?

timkirby, 2020-09-16:
Thanks Andreas! I’m on a hosted instance, so I don’t think I can inspect the value in the file, unfortunately.

I’m sure you’re right, it’s some sort of initialisation issue. The default screen display for the measurement parameter showed KM, but I’m guessing that was not loaded until I made another change? Could it be worth defaulting a value into the DB, might that work?

Tim

magicbug, 2020-09-16:
Weird, just testing here, remembering this will only change the measurement value when logged in and when i change the km/m value in profile then go back and click on a QSO the data changes.

magicbug, 2020-09-16:
OK, just tested on Tims setup with my own account

- Selected Miles
- Saved
- Logbook -> clicked on a QSO showed miles
- Edit Profile
- Changed to KM
- Saved
- Back to the logbook -> so popup and shows correctly for me.

Be interesting if you can try the same method Tim

magicbug, 2020-09-16:
Unless we are talking about a different area of Cloudlog?

timkirby, 2020-09-16:
Hi both,

Sorry for the delay - been outside.

Your method works fine, Pete. Also, there is no problem now and I can go back and forth from KM to Mi as required.

I *think* it’s a one off on first run and this is what I think had happened.

I looked in my account and it was displaying kilometre as measurement preference. Great, I thought and seeing it was set that way, I didn’t change it or hit save the account parameters. Went to look at the Edit QSO dialog and it was displaying in miles. Well, that’s odd I thought - because the account details had said KM - and that was when I messaged you, Pete! 

Had I hit enter when I first went into account parameters, I think it would would have worked straightaway, because I would have then saved the default value from the screen (KM) into the database.

So if there’s a ‘feature’ I think it’s only evident on first access of the account details screen after this change, and that ‘feature' is that the Measurement Preference displays as KM when the system is still operating in miles.

I hope I’ve explained that ok!

Thank you both for your help.

magicbug, 2020-09-16:
Yep, agree, need to have a blank select option for when the table rows not been filled in. It's definitely a bug.

magicbug, 2020-09-16:
Thanks for reporting this Tim, now fixed within the codebase

closed 2020-09-16 by magicbug