magicbug/Cloudlog #622: PHP error in /application/models/Dxcc.php on line 171

lu1aax, 2020-09-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When trying to create a station profile

I get this:

Parse error: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING) in /srv/www/lu1aax.taslik.net/Cloudlog-master/application/models/Dxcc.php on line 171
A PHP Error was encountered
Severity: Parsing Error

Message: syntax error, unexpected 'list' (T_LIST), expecting identifier (T_STRING)

Filename: models/Dxcc.php

Line Number: 171

magicbug, 2020-09-17:
Last time I saw a T List error it was PHP5 can you confirm PHP version is over the minimum php7 as per the readme

magicbug, 2020-09-18:
Duplicate post fixed by using php7

closed 2020-09-18 by magicbug