magicbug/Cloudlog #623: Previous QSOs with Callsign

M0TZO, 2020-09-18:
|In the the pop up QSO details box, would it be possible to show Previous QSOs?

Also on the Homepage &/or Logbook pages maybe a symbol (e.g. 2M0SQL + ) to show there are previous QSOs?

I've worked a few recently that Gridtracker is saying Previous bands / modes worked,

magicbug, 2020-09-18:
I've been thinking about a More Details full page for worked callsigns that would probably be the better solution rather than adding it to the QSO details popup.

This would show everything in the system about the callsign, past QSOs, QSL statuses etc.

magicbug, 2020-09-18:
Currently won't be added any time in the next couple of weeks, I need to get my focus back on pdf label generation for a bit.

closed 2020-12-11 by magicbug