magicbug/Cloudlog #630: Create a Global Configurations Options stored in MySQL table config or similar

magicbug, 2020-09-24:
Its become apparent going forward Cloudlog will need Global Configuration options that go out across the whole installation., this could be things like QRZ logins, CSS Stylesheet (whether you have light or dark, default date formats etc etc.

This would only be assigned to "Admins" which will become more useful when I finally get round to making cloudlog better for multi-user, coming soon with the creation of Logbooks #609 to sit in front of Station Profiles, will also help us move away from static configuration files!

m0pcb, 2020-09-24:
This is of particular value to users of hosted instances who cannot rely on
editing config files. Presenting these options to a UI page will also be
an important feature. :)

73, Iain AD5XI / M0PCB

magicbug, 2020-12-10:
This feature is now coming together I've added a migration file that creates an options table similar to what WordPress uses and an option for yes/no to auto load the option.

Next up is coding to automatically load the options and start using them around Cloudlog.

closed 2020-12-11 by magicbug