magicbug/Cloudlog #631: Wrong frequency from SatPC-32 Interface (Only QO-100)

HB9WDF, 2020-09-28:
**Describe the bug**
It's only with the SatPC-32 Interface
It's only with the QO-100 Satellite
The live shown frequency in the log is correct.
After the saving of the qso, there is a wrong value in the log

**To Reproduce**
I use SatPC32 Version 12.8d with the current version of Cloudlog and the SatPC-32 Interface
For QO-100, I use the following definition in the Doppler.sqf: QO-100,10489740,2400240,USB,USB,NOR,0,0,CW/SSB

1. Choose QO-100 for the QSO
2. Start with "Live QSO" in Cloudlog
3. The Satellite and the Frequencys are correct (TX: 2400240000 / RX: 10489739998)
4. After saving the qso, open the qso in the Log in edit mode
5. The Showing value is: -1894727296 

**Expected behaviour**
Save the right frequency

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - Browser All Browsers (Chrome, Firefox, Edge)
 - Current Versions

**Smartphone (please complete the following information):**
- Not tested yet

**Additional context**
There is no problem with other satellites in VHF UHF

magicbug, 2020-09-28:
Interesting, can i see a screenshot of the desktop satpc<>cloudlog app too please

magicbug, 2020-09-28:
I've just tried this my end, and can't replicate this using the latest codebase and your doppler line

magicbug, 2020-09-28:
Are you running the latest codebase or one from a while ago?

HB9WDF, 2020-09-29:
Hi Peter

 

I updated the issue with the screenshot of SatPC-32 and the Cloudlog interface.

I (hope) hat I running the latest codebase. Last one was a few days ago (via git pull)

 

Cloudlog is installed on a raspi.

 

Maybe it’s not the codebase…

What can I do wrong during installation on a raspi, that I got the wrong data?

 

I have an indication for my proposition: I have still the old blue “pins” on the map. Not the new red dot’s. And I can’t find a line in the config, where I can change that.

Maybe, I did a mistake during the installation process? 

 

Thank you for your support Peter.

 

73s Michael

Hb9wdf

magicbug, 2020-09-29:
Very weird, blue pins indicate update isn't working via git pull that is changed by default.

on the command line when you "type git pull" what do you get if there's errors, can you type "git stash" then do "git pull"

I have a feeling old version!

HB9WDF, 2020-09-29:
Hi Peter
That I got with git pull today
And with "git stash" and git stash show the following:
73s Michael

magicbug, 2020-09-29:
Missed the content however did it update Cloudlog?

DG7YEO, 2020-10-03:
Have the same issue on my QO 100 QSOs, but I don't use SatPC-32 Interface.

If I edit a QSO I did via QO-100 I see a negative value in the Frequency.
When I added the Frequency in live QSO I add ie 2400000000 when added all other and save the QSO and later would like to edit the QSO Data I get a negative value in the Frequency, see picture below.

[image]

73
DG7YEO

magicbug, 2020-10-12:
@DG7YEO Sorry for delay been unwell, can you confirm if this is on the last version of the codebase on github, I'm guessing its latest, but i genuinely cant replicate the issue so be good to know which version of the codebase is dragging this up.

DG7YEO, 2020-10-12:
Just made a git pull to be sure, is there anything else need to be done to make sure I'm up to date? 
After that I entered a test QSO and still the same problem. I enter 2400000000 into Frequency but when I got with Edit into the log entry Frequency show -1894792296. 

73
DG7YEO

HB9WDF, 2020-10-15:
Hi Peter
I made a new install of Cloudlog on my Raspi. Now the update function works fine. It was a wrong installation, that was my mistake. I'm still learning with Linux ;-)
But unfortunately I have still the Issue with wrong value in the Frequency. The Issue is in "Live QSO" and also in "Post QSO". I tryed with the SatPC32 Interface and also with "none" in the Radio selection.
73s Michael

HB9WDF, 2020-10-18:
I played a little bit around. I figured out, that the highest value is 2147483647. After that, it goes negativ.
So I asked google. It's look like, that is a common issue in MySQL. Maybe the following can help.
https://stackoverflow.com/questions/10255724/incorrect-integer-2147483647-is-inserted-into-mysql.

HB9WDF, 2020-10-24:
Now I think I have found the main Issue:
2147483647 is the highest number in a 32 Bit system. My Raspi has an ARMv7 Proccessor and I think, it's running only with 32 Bit.
In your online demo, the higher frequencies wirks fine.

HB9WDF, 2020-10-25:
So, today I installed Ubuntu 64 Bit on a spare barebone pc. byebye win10 ;-)
After that, I installed a LAMP server and Cloudlog. 
Now, it works perfect with the frequency higher than 2147483647.
If somebody have the same issue, then it must be a 32 Bit system. (Older Raspi work with 32 Bit...)

closed 2020-10-25 by HB9WDF