magicbug/Cloudlog #632: Feature: Add Translations to Cloudlog

magicbug, 2020-09-28:
With Cloudlogs users being international getting lots of comments about making it translatable which is do able using https://codeigniter.com/userguide3/libraries/language.html just requires building files and changing the views which does add work but long term probably worthwhile.

I'm guessing this is of interest to everyone and who would like to get involved

phl0, 2020-09-28:
I like the idea. Could provide German translations. 😎

magicbug, 2020-09-28:
Guess best idea is to start per section of Cloudlog, this is likely to take a while

magicbug, 2020-09-28:
Decided to have a play and do the front dashboard page its now committed and you can see the language file at https://github.com/magicbug/Cloudlog/blob/master/application/language/english/general_words_lang.php

cyber-atomus, 2020-09-28:
Ha, posted on tweetear an idea and it's already alive! Dang that was fast! Reading abt codeigniter took me a while, but I did almost the same job as @magicbug. It's there: https://github.com/cyber-atomus/Cloudlog - also translated some stuff to Polish already...

magicbug, 2020-09-29:
Works for myself, PRs are accepted, I only did it roughly to see how it could work 👍🏻

phl0, 2020-11-25:
I did add basics for German with https://github.com/magicbug/Cloudlog/pull/720

closed 2020-12-11 by magicbug