magicbug/Cloudlog #635: Days QSO Feature (AndreasK79)

M0TZO, 2020-09-30:
Sorry to be a pain, as was closed, but the start dates for me don't match on QSO Streak feature.

Current streak shows 80 days _2020-04-25 to 2020-09-30_, but should be _2020-07-13 to 2020-09-30_

Thanks in Advance, sorry for delay in reporting.

AndreasK79, 2020-09-30:
Nothing to be sorry about. I will take another look at the code.

AndreasK79, 2020-09-30:
#636 added. This should fix the date issue.

M0TZO, 2020-10-01:
Yep, looks good now. Thanks.

M0TZO, 2020-10-01:
Odd, was correct July date yesterday, now reverted. Hopefully will show up in the updates soon. :)

magicbug, 2020-10-01:
Just merged, should be available to grab

M0TZO, 2020-10-03:
One last issue. I think it includes the first date which should be day zero. Today should be day 82 when I've worked someone, but already shows 82 and will be 83 when I have a QSO, if that makes any sense.

AndreasK79, 2020-10-03:
I'm not sure I follow. Day 1 is day 1. It is the consecutive days with QSOs. If you have 5 following days with at least 1 QSO, that is a streak of 5.

M0TZO, 2020-10-15:
Some would count as day Zero, I'm happy it's working so ignore me, Up to 95 days now :)

closed 2020-10-28 by M0TZO