magicbug/Cloudlog #637: QRZ Callsign Lookup

myzinsky, 2020-10-02:
I observed that the QRZ callsign lookup does not work anymore, even its configured correctly and it was working before. Has something changed in this code part?

dg9vh, 2020-10-02:
Can't confirm, on my side it's working as ever... maybe on your side something changed? XML-account expired?

myzinsky, 2020-10-02:
Hi Kim, negative, with the CloudLogOffline app qrz.com still works. Just checked again the configuration in php, it also looks fine. Therefore, I'm asking :)

magicbug, 2020-10-02:
Working ok here, I'd suggest logging out and in just in case of something is wonky with the session key from qrz

myzinsky, 2020-10-02:
@magicbug The QSO upload in the log still works. But the call lookup is not triggered for live and post QSOs. Strange... after a logout and login it works again... seams that you had the right idea :-)

closed 2020-10-02 by myzinsky