magicbug/Cloudlog #641: Active station profile based on user

dg9vh, 2020-10-04:
We are planning to use Cloudlog as log for our clubstation activity next year (aniversary of our local radio club). So we want to create one log with several users having their own station profile (because they work from different places etc.).

Actually there is one central station profile active for each user, so actually each user has to check station profiles and activate his personal profile for being active.

It would be better to have the active station profile associated with the user-login. Means to have a field within the user-table that contains the station profile for the user.

With this little modification we have the possibility to create logins and get them their personal profile without having the user to change this profile on each login.

If this would be accepted by the broad mass here, I will start to modify the profile-handling to work this way.

dg9vh, 2020-10-06:
After no negative comments happened for two days I will start creating a concept for development now...

magicbug, 2020-10-09:
Not so much no opinion life's been busy I'll take a look at the PR when I have time.

dg9vh, 2020-10-10:
No problem

AndreasK79, 2020-11-29:
@dg9vh I see you closed the PR. I never commented on this, as I didn't really have an opinion on this. 

Would this change affect anything else in the log or code? Would a rewrite be needed several places to get the active profile?

dg9vh, 2020-11-29:
I closed it as it seems not to be accepted by peter for long long time now... so I thought it would not be necessary or something other ways would be planned...

There where a migration file on for modifying database and some changes on the user-profile... It affected several files so I think it will be ok if it was thrown away.

closed 2020-12-11 by magicbug